newton-physics/newton #48: Add shape flags

eric-heiden, 2025-05-07:
Maybe 16 bits reserved for the user

eric-heiden, 2025-05-07:
I think we can still query the flags from `find_shape_contact_pairs`, just need to use binary checks to figure out the flags:
```py
if self.shape_flags[shape_a] & SHAPE_FLAG_COLLIDE_SHAPES:
   ...
```
Having the flags as Warp array on the `Model` would also allow us to use them in the new collision library.

vreutskyy, 2025-05-07:
Hmm. But it's `Model.find_shape_contact_pairs`. and it's a Warp array there, not a Python list. What do I miss?

eric-heiden, 2025-05-07:
Oh indeed, I missed that! But I think it can be moved to `ModelBuilder`, or at least get refactored that way until the Warp arrays `shape_contact_pairs` and `shape_ground_contact_pairs` are constructed.

vreutskyy, 2025-05-07:
There're 2 more places where it tries to read that array. In MuJoCoSolver.convert_to_mjc.add_geoms and SimRenderer.populate

eric-heiden, 2025-05-13:
Fixed in https://github.com/newton-physics/newton/pull/69

closed 2025-05-13 by eric-heiden