newton-physics/newton #55: Modelbuilder Performance

adenzler-nvidia, 2025-05-07:
Simulating example_humanoid.py with 8k envs spends 80s in model building, compared to 7 seconds for the simulation.

A quick inspection tells that 60seconds are spent in modelBuilder.add_builder(). The obvious case to look at is all the extend/append calls to python lists.

There are a few ways this could be fixed, but ultimately this needs a few experiments to figure out what's best. We don't want to break things for heterogeneous environments but we should make sure we can be fast for homogeneous environments where we basically know upfront how much memory we need. Maybe we need to switch to numpy arrays.

eric-heiden, 2025-05-07:
One idea suggested by @nvlukasz is to switch all the Python lists in `ModelBuilder` to numpy arrays.

Then we could also experiment with an extra argument to `ModelBuilder.add_builder()` to replicate the other builder N times (potentially with some spacing applied). This would also be much faster using numpy functions.

preist-nvidia, 2025-07-09:
@eric-heiden - since cloning is a KPI for the alpha release, this may be an alpha issue rather than beta.

Kenny-Vilella, 2025-07-24:
For reference,  it takes around 2min in example_mujoco.py to initialize 8k H1.
I didn't look at trace, but it is probably also mainly the model builder.