newton-physics/newton #93: Bug in setting start pose of a rigid body.

gyeomannvidia, 2025-05-13:
There are two ways to set the start pose of a rigid body. 

The 1st method is to set the start pose with ModelBuilder.add_body(origin = myStartPose)

The 2nd method is to set the pose of the free joint with rbody_builder.joint_q[-7:] = [myStartPose[0], myStartPose[1], .......myStartPose[6]]

There are two issues:
1. The mujoco solver ignores method 1 above.
2. The xpbd solver ignores method 2 above.

The attached example illustrates the issue.  I had to rename the file to *.txt to get it to upload so it needs to renamed with the py extension.

[example_boxonplane.txt](https://github.com/user-attachments/files/20191412/example_boxonplane.txt)

xpbd and mujoco solvers can be toggled by changing the parameter: use_xpbd

The two methods of setting the start pose are found in lines 116 to 132 and can be commented out as required for testing.

eric-heiden, 2025-05-13:
This is connected to https://github.com/newton-physics/newton/issues/43. We should create free joints for those bodies.
When creating a free joint we can use the child body's `body_q` as initialization for its `joint_q` value.
In the direction of `joint_q` -> `body_q`, we could probably invoke `eval_fk` when calling `ModelBuilder.finalize()`.

gyeomannvidia, 2025-05-14:
If we move to exclusively using joint_q then we should remove the origin parameter from add_body. 

The difficulty for a noob user like me is that there appear to be 2 methods to set the start pose of a rigid body but each method is specific to the chosen solver.