newton-physics/newton #95: Bug with plane collisions

nvlukasz, 2025-05-13:
When importing a ground plane from USD, the collisions aren't correct. The plane seems to import fine and the collision groups look alright. To repro, remove the .txt extensions from the attachments.

There might actually be more than one issue here.

1. With `self.model.ground = False`, the ants fall through the imported ground plane. It happens with both the MuJoCo and XPBD solvers.

2. With `self.model.ground = True` and the MuJoCo solver, the ants collide with the builtin ground plane, except for one leg:

[image]

3. The ants don't get imported at the correct height. In USD, they are at z=1, but they spawn at z=0 by default. This means that I need to manually specify the xform for the ants when calling `add_builder()`.

[repro_ant_plane.py.txt](https://github.com/user-attachments/files/20193291/repro_ant_plane.py.txt)
[repro_ant_plane.usda.txt](https://github.com/user-attachments/files/20193292/repro_ant_plane.usda.txt)

nvlukasz, 2025-05-16:
Updated repro script:
```Python
# SPDX-FileCopyrightText: Copyright (c) 2025 The Newton Developers
# SPDX-License-Identifier: Apache-2.0
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
# http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import numpy as np
import warp as wp

import newton
import newton.core.articulation
import newton.examples
import newton.utils

class Example:
    def __init__(self, stage_path="example_ant.usd", num_envs=8):
        self.num_envs = num_envs

        fname = "repro_ant_plane.usda"
        # fname = "/work/src/newton/newton/examples/assets/ant_prototype.usd"
        # fname = "/work/src/newton/newton/examples/assets/ant_prototype.usda"
        prototype_path = "/World/envs/env_0"
        path_pattern = "/World/envs/env_{}"

        up_axis = "Z"

        # load only global geometry (exclude prototype env)
        builder = newton.ModelBuilder(up_axis=up_axis)
        newton.utils.parse_usd(
            fname,
            builder,
            ignore_paths=[prototype_path],
            collapse_fixed_joints=True,
            joint_ordering="dfs",
        )

        # load just the prototype env
        prototype_builder = newton.ModelBuilder(up_axis=up_axis)
        newton.utils.parse_usd(
            fname,
            prototype_builder,
            root_path=prototype_path,
            collapse_fixed_joints=True,
            joint_ordering="dfs",
        )

        env_offsets = newton.examples.compute_env_offsets(num_envs, env_offset=(3.0, 3.0, 0.0), up_axis=up_axis)

        # clone the prototype with updated paths
        for i in range(num_envs):
            body_start = builder.body_count
            shape_start = builder.shape_count
            joint_start = builder.joint_count
            articulation_start = builder.articulation_count

            # BUG? the ants should be at z=1 in USD, but they spawn at z=0 unless we specify the xform explicitly
            xform = wp.transform(env_offsets[i] + np.array((0.0, 0.0, 1.0)), wp.quat_identity())
            builder.add_builder(prototype_builder, xform=xform)

[... 117 lines cut ...]
    parser.add_argument(
        "--stage_path",
        type=lambda x: None if x == "None" else str(x),
        default="example_ant.usd",
        help="Path to the output USD file.",
    )
    parser.add_argument("--num_frames", type=int, default=1200, help="Total number of frames.")
    parser.add_argument("--num_envs", type=int, default=4, help="Total number of simulated environments.")

    args = parser.parse_known_args()[0]

    with wp.ScopedDevice(args.device):
        example = Example(stage_path=args.stage_path, num_envs=args.num_envs)

        for _ in range(args.num_frames):
            example.step()
            example.render()

            # # !!! throttle the frame rate
            # import time
            # time.sleep(0.1)

        if example.renderer:
            example.renderer.save()
```

eric-heiden, 2025-06-17:
This bug has been fixed.