newton-physics/newton #139: Newton Model -> MoJuCo Model verification and documentation

adenzler-nvidia, 2025-05-22:
The goal of this task is to verify and test for all the newton parameters that they end up in the right place in the mujoco model. Along with this, we can add support for runtime model updates for the fields that allow this in MuJoCo.

These are the requirements for all sub-tasks:
* It is clearly documented how a newton model parameter is translated to the mujoco model. This includes units and defaults.
* Each newton model field has a test that verifies the functionality in the mujoco solver.
* For each field that is randomizable, there is a test that verifies that the per-environment values are correctly translated to the mujoco worlds at solver initializations.
* For each field that is randomizable, there is a test that verifies that model updates for multiple envs at runtime are correctly translated to mujoco.

This will allow a future workstream that will make us not dependent on compiling a mujoco model anymore, but allow us to directly build the mujoco model from the newton model.

For each newton model parameter, the following needs to be done:
* Figure out what fields in the mujoco model are being influenced by this, and document how they are filled.
* in case the field is randomizable in mujoco, implement a conversion function that converts from the newton model to the mujoco model.
* hook up that function for both runtime updates and initialization
* add tests.

There is a draft PR here that already does this for inertia and body mass and adds some of the boilerplate: https://github.com/newton-physics/newton/pull/116

This one will be cleaned up and should be the basis for this work.

adenzler-nvidia, 2025-05-22:
This is the list of randomizable mujoco model fields as of right now: 
https://github.com/adenzler-nvidia/newton/blob/dev/adenzler/first-part-model-expansion/newton/solvers/solver_mujoco.py#L1428

preist-nvidia, 2025-08-28:
Reopened and re-tagged for beta, so we can continue on this, please update description @adenzler-nvidia

preist-nvidia, 2025-10-07:
Closing for Beta 1 scope, will create new epic for Beta 2 scope.

preist-nvidia, 2025-10-07:
Follow-up is #900