newton-physics/newton #147: Fix `body_iquat` computation in MuJoCoSolver

eric-heiden, 2025-05-23:
`update_body_inertia_kernel` currently doesn't update the `mjw_model.body_iquat` field. Applying the usual up-axis handling to that rotation also didn't fix it so far.

adenzler-nvidia, 2025-07-16:
I'm trying to figure out how to interpret the inertial frames in newton. Is the newton inertia always in the body frame, or is there a separate inertial frame?

If it's always body frame, we can just omit writing iquat because it will be inferred by us specifying fullInertia. If I understand the docs correctly.

adenzler-nvidia, 2025-08-28:
This is the code in MuJoCo that goes from fullinertia to diagonal + quat: https://github.com/google-deepmind/mujoco/blob/b1c94c37618eb10df0ce1df0093fc455bc632959/src/user/user_util.cc#L785

adenzler-nvidia, 2025-08-29:
fix in #685

closed 2025-08-29 by eric-heiden