newton-physics/newton #174: CUDA Error 900 during CUDA Graph Capture in example_cloth_self_contact.py

YoruCathy, 2025-05-28:
When running the example_cloth_self_contact.py example, the program crashes due to a CUDA error during graph capture.

**Environment:**
- Newton version: main branch, commit f9403323a2be007cd857551f953818a240f230d2
- Warp version: 1.8.0.dev20250522
- Python: 3.11 (I'm using the venv in this repo)
- GPU: NVIDIA GeForce RTX 4090
- CUDA Toolkit: 12.8, driver 12.2
- OS: 20.04

**Steps to Reproduce:**
uv run newton/examples/example_cloth_self_contact.py

**Observed Behavior:**
The script fails with the following error:
```
(newton-physics) 🐱 [cathy@mountain-lion] newton [main] ⚡️ uv run newton/examples/example_cloth_self_contact.py 

Warp 1.8.0.dev20250522 initialized:
   Git commit: f9403323a2be007cd857551f953818a240f230d2
   CUDA Toolkit 12.8, Driver 12.2
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 4090" (24 GiB, sm_89, mempool enabled)
   Kernel cache:
     /home/cathy/.cache/warp/1.8.0.dev20250522
Warp UserWarning: The graph is not optimizable anymore, terminated with a max/min ratio: 2.0 without reaching the target ratio: 1.1
Module newton.core.graph_coloring 3c9ae71 load on device 'cpu' took 2.60 ms  (cached)
Module newton.solvers.solver_vbd c2aad55 load on device 'cpu' took 1821.75 ms  (compiled)
Module newton.collision.collide 51ed862 load on device 'cuda:0' took 1.56 ms  (cached)
Module __main__ 166c637 load on device 'cuda:0' took 618.15 ms  (compiled)
Module warp.render.render_opengl 25e96b8 load on device 'cuda:0' took 497.47 ms  (compiled)
Module warp.utils 6a774bf load on device 'cuda:0' took 239.07 ms  (compiled)
Module warp.math 8f92a51 load on device 'cuda:0' took 102.36 ms  (compiled)
Module newton.core.types 8f92a51 load on device 'cuda:0' took 102.47 ms  (compiled)
Module newton.core.spatial 8f92a51 load on device 'cuda:0' took 106.19 ms  (compiled)
Module newton.core.articulation 3dd9ac3 load on device 'cuda:0' took 0.93 ms  (cached)
Module newton.core.graph_coloring 3c9ae71 load on device 'cuda:0' took 226.68 ms  (compiled)
Module newton.core.inertia 4136106 load on device 'cuda:0' took 0.74 ms  (cached)
Module newton.solvers.solver 00c0bde load on device 'cuda:0' took 0.68 ms  (cached)
Module newton.core.particles 1812946 load on device 'cuda:0' took 241.09 ms  (compiled)
Module newton.solvers.solver_euler 102b0b8 load on device 'cuda:0' took 0.96 ms  (cached)
Module newton.solvers.solver_featherstone 0a91b6d load on device 'cuda:0' took 1.09 ms  (cached)
Module warp.render.utils 5eaf2f2 load on device 'cuda:0' took 212.37 ms  (compiled)
Module newton.utils.render 98d6c0d load on device 'cuda:0' took 190.14 ms  (compiled)
Module newton.utils 8f92a51 load on device 'cuda:0' took 103.85 ms  (compiled)
Module newton.solvers.solver_mujoco f33068b load on device 'cuda:0' took 0.94 ms  (cached)
Module newton.solvers.solver_vbd c2aad55 load on device 'cuda:0' took 1666.66 ms  (compiled)
Module newton.solvers.solver_xpbd ee59012 load on device 'cuda:0' took 1.34 ms  (cached)
Warp CUDA error 900: operation not permitted when stream is capturing (in function cuda_load_module, /builds/omniverse/warp/warp/native/warp.cu:3913)
Warp CUDA error: Loading module failed
Module newton.solvers.solver_vbd e3fe310 load on device 'cuda:0' took 1637.89 ms  (error)
Traceback (most recent call last):
  File "/home/cathy/Workspace/newton/newton/examples/example_cloth_self_contact.py", line 238, in __init__
    self.integrate_frame_substeps()
  File "/home/cathy/Workspace/newton/newton/examples/example_cloth_self_contact.py", line 263, in integrate_frame_substeps
    self.solver.step(self.model, self.state0, self.state1, self.control, None, self.dt)
  File "/home/cathy/Workspace/newton/newton/solvers/solver_vbd.py", line 2117, in step
    self.simulate_one_step_with_collisions_penetration_free(model, state_in, state_out, dt, control)
  File "/home/cathy/Workspace/newton/newton/solvers/solver_vbd.py", line 2229, in simulate_one_step_with_collisions_penetration_free
    self.collision_detection_penetration_free(state_in, dt)
  File "/home/cathy/Workspace/newton/newton/solvers/solver_vbd.py", line 2355, in collision_detection_penetration_free
    wp.launch(
  File "/home/cathy/Workspace/newton/.venv/lib/python3.11/site-packages/warp/context.py", line 5853, in launch
    module_exec = kernel.module.load(device, block_dim)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cathy/Workspace/newton/.venv/lib/python3.11/site-packages/warp/context.py", line 2384, in load
    raise Exception(f"Failed to load CUDA module '{self.name}'")
Exception: Failed to load CUDA module 'newton.solvers.solver_vbd'

[... 106 lines cut ...]
  File "/home/cathy/Workspace/newton/.venv/lib/python3.11/site-packages/warp/context.py", line 6064, in launch_tiled
    return launch(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cathy/Workspace/newton/.venv/lib/python3.11/site-packages/warp/context.py", line 5853, in launch
    module_exec = kernel.module.load(device, block_dim)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cathy/Workspace/newton/.venv/lib/python3.11/site-packages/warp/context.py", line 2384, in load
    raise Exception(f"Failed to load CUDA module '{self.name}'")
Exception: Failed to load CUDA module 'mujoco_warp._src.smooth._tile_cholesky'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/cathy/Workspace/newton/newton/examples/example_humanoid.py", line 145, in <module>
    example = Example(stage_path=args.stage_path, num_envs=args.num_envs)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cathy/Workspace/newton/newton/examples/example_humanoid.py", line 101, in __init__
    with wp.ScopedCapture() as capture:
  File "/home/cathy/Workspace/newton/.venv/lib/python3.11/site-packages/warp/utils.py", line 1516, in __exit__
    self.graph = wp.capture_end(device=self.device, stream=self.stream)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cathy/Workspace/newton/.venv/lib/python3.11/site-packages/warp/context.py", line 6353, in capture_end
    raise RuntimeError(f"CUDA graph capture failed. {runtime.get_error_string()}")
RuntimeError: CUDA graph capture failed. Warp CUDA error 900: operation not permitted when stream is capturing (in function cuda_load_module, /builds/omniverse/warp/warp/native/warp.cu:3913)
```

YoruCathy, 2025-05-28:
Nvm I figured my cuda is somehow still using 11.8, and I updated to 12.8 w/ newer driver. It's working now.

closed 2025-05-28 by YoruCathy

Zvch17, 2025-07-31:
Hi, I believe I’ve run into the same issue. I’m running `example_cloth_hanging.py`, and the error I get is:
```
RuntimeError: CUDA graph capture failed. Warp CUDA error 900: operation not permitted when stream is capturing (in function wp_cuda_load_module, warp.cu:3944)

```
Here’s the full initialization output from my run:
```
Warp 1.9.0.dev20250730 initialized:
   Git commit: 8d1c24401b94114cdda35b7a68d63711cabf7d45
   CUDA Toolkit 12.8, Driver 12.2
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 4090" (24 GiB, sm_89, mempool enabled)
   Kernel cache:
     /home/zach/.cache/warp/1.9.0.dev20250730

```
I’d like to ask:

How exactly did you verify that CUDA 11.8 was being used at runtime, despite the Warp log showing Toolkit 12.8?

What steps did you take to actually force Warp to use CUDA 12.1 and resolve the issue?