newton-physics/newton #241: Instability and Non-zero status in Cholesky

nvlukasz, 2025-06-17:
[repro_cartpole_instability.zip](https://github.com/user-attachments/files/20783748/repro_cartpole_instability.zip)

To repro, unzip and run script.

We're applying forces to the cart based on its position. It doesn't take long before the cartpoles start exploding and we get a flood of
```
Non-zero status in Cholesky factorization, got 1
```

Seems like energy keeps increasing. We are also seeing these types of errors in more complex tasks, like training humanoids.

nvlukasz, 2025-06-18:
[repro_cartpole_instability_det.zip](https://github.com/user-attachments/files/20796768/repro_cartpole_instability_det.zip)

Here's a more deterministic variant that doesn't randomize the initial poses. This reliably explodes on my machine after about 20 seconds of sim time, but it may vary by GPU etc. It can be tricky to find an initial pose that guarantees the explosion, so it's worth trying some randomized poses if the default ones don't repro.

adenzler-nvidia, 2025-06-30:
The bugs around error messaging are resolved in warp - any remaining occurrences of these look all like legitimate solver explosions due to erratic user inputs.

closed 2025-06-30 by adenzler-nvidia