newton-physics/newton #410: Signed distance issue in create_soft_contacts

CliffieVanR, 2025-07-15:
Hi,
In create_soft_contacts, the distance 'd' returned by the various low-level functions is signed, but the acceptance criteria at line 749 (kerens.py) appears to assume it's always positive. Is this intentional, or ought there to be an abs(d) in there?

Thanks in advance,
Cliff

adenzler-nvidia, 2026-10-14:
Thanks for the question, Cliff. The signed distance is intentional. `create_soft_contacts` is a candidate filter: `d < margin + radius` accepts a particle whose surface is within `margin` of the shape surface or already penetrating it. Penetration gives `d < 0`, so it always passes; using `abs(d)` would wrongly reject deeply penetrating particles. The contact point `x_local - n * d` also relies on the sign to project back onto the surface, and the solvers recompute the actual penetration from the stored point and normal.

One related limitation for mesh shapes: the closest-point query only searches within `margin + radius` of the surface, so a particle deeper than that inside a closed mesh gets no contact. That's a separate search-radius issue rather than a sign problem.

Closing; feel free to reopen with a repro if you see a case that misbehaves.

closed 2026-10-14 by adenzler-nvidia

CliffieVanR, 2026-10-15:
Hi Alain, of course! Many thanks for the clarification.