newton-physics/newton #556: Unitree Go1 walking issue in Newton

sunqiang188, 2025-08-15:
I used the pretrained policy from Isaac Lab, and then rewrote it based on the Anymal C walking example in Newton. However, the Go1’s walking posture is incorrect.Here are the video and the code:

https://github.com/user-attachments/assets/1a66e546-33a6-4f52-9df0-47492c263df4

[go1_walk_example.zip](https://github.com/user-attachments/files/21790196/go1_walk_example.zip)

preist-nvidia, 2025-09-01:
Hi @sunqiang188 - we are still in alpha development so it can be that you hit a bug that has been fixed in the meantime. The Anymal example had an issue that we fixed, so it may be that the bug you hit is fixed now, can you please check?

preist-nvidia, 2025-10-07:
Closing for now until we hear back from you @sunqiang188 .

closed 2025-10-07 by preist-nvidia