newton-physics/newton #585: example_anymal_c_walk_physx_policy.py broken

erwincoumans, 2025-08-19:
newton-physics             0.1.1a2 
mujoco                     3.3.6.dev796509754
mujoco-warp                0.0.1 

Running example_anymal_c_walk_physx_policy.py  reports this error:

```
(roger_conda310) erwin@ecoumans-desktop:~/dev/roger/newton/newton$ python examples/example_anymal_c_walk_physx_policy.py 
Warp 1.9.0.dev20250819 initialized:
   Git commit: 50da2e5729073dd52d3a6c982754aea7f9018976
   CUDA Toolkit 12.8, Driver 12.9
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 5090" (31 GiB, sm_120, mempool enabled)
   Kernel cache:
     /home/erwin/.cache/warp/1.9.0.dev20250819
Module newton._src.geometry.inertia 50ad00d load on device 'cuda:0' took 0.62 ms  (cached)
Module validate_and_correct_inertia_kernel_4d850460 4d85046 load on device 'cuda:0' took 0.36 ms  (cached)
Module count_contact_points_84692ae6 84692ae load on device 'cuda:0' took 0.34 ms  (cached)
Module newton._src.sim.graph_coloring 1fe8518 load on device 'cpu' took 1.17 ms  (cached)
Traceback (most recent call last):
  File "/media/erwin/ubuntu_extra/dev/roger/newton/newton/examples/example_anymal_c_walk_physx_policy.py", line 235, in <module>
    example = Example(stage_path=args.stage_path, headless=args.headless)
  File "/media/erwin/ubuntu_extra/dev/roger/newton/newton/examples/example_anymal_c_walk_physx_policy.py", line 145, in __init__
    self.solver = newton.solvers.SolverMuJoCo(self.model)
  File "/media/erwin/ubuntu_extra/dev/roger/newton/newton/_src/solvers/mujoco/solver_mujoco.py", line 1186, in __init__
    self.convert_to_mjc(
  File "/media/erwin/ubuntu_extra/dev/roger/newton/newton/_src/solvers/mujoco/solver_mujoco.py", line 2203, in convert_to_mjc
    assert len(base_shapes) == len(body_shapes)
AssertionError
```

erwincoumans, 2025-08-20:
FYI: changing the solver to
'''
self.solver = newton.solvers.SolverFeatherstone(self.model)
'''
make the example work, and walk for a little while.

eric-heiden, 2025-08-22:
This example is working again now using the SolverMuJoCo.

closed 2025-08-22 by eric-heiden