newton-physics/newton #745: investigate issues with parallel linesearch on MuJoCo humanoid

adenzler-nvidia, 2025-09-05:
For some reason enabling parallel linesearch leads to very bad convergence once the humanoid hits the ground. I traced it down to the control inputs but couldn't figure out what exactly the problem is, so I'm deferring an actual investivation to this ticket to make progress on #738.

Scaling the range of the control noise down helps significantly. Adding armature and effort limits didn't seem to have an effect. Taking the config from the robot example made the problem even worse.

adenzler-nvidia, 2025-09-10:
Looks like the KPI* benchmarks didn't take that change well either - I can see:
* clear regression on KPIAnt, somewhat expected because the numbers were flat for the fast version. Probably also convergence issues.
* KPIG1 with a small regression for 4k, doesn't run anymore for 8k envs
* KPIH1 with a regression for 8k envs

Kenny-Vilella, 2025-09-16:
I didn't see any issue with the KPI benchmarks on trunk.
And everything is working as expected when changing the way to measure time.

One of problem is that after this change, we will not be consistent with the way we measure the step time across all the benchmarks.
But as long as we use in the benchmark example scripts that are meant to be "tutorial", we may have such issues.

PR #822 (2025-10-02): Improve time measurement in benchmark
## Description
- Changed the way time is calculated in the Fast benchmark to match the KPI benchmark
- Changed the way we measure the benchmark time to capture only the simulation time and exclude the action setting.

This will fix #745 and is also related to #652.

## Newton Migration Guide

Please ensure the migration guide for **warp.sim** users is up-to-date with the changes made in this PR.

- [ ] The migration guide in ``docs/migration.rst`` is up-to date

## Before your PR is "Ready for review"

- [ ] Necessary tests have been added and new examples [... cut]
closed 2025-10-02 by Kenny-Vilella