newton-physics/newton #838: Inertia issue in robot_anymal_c_walk

lenroe-nv, 2025-09-26:
Looking at the example `robot_anymal_c_walk`, there might be something wrong with the base inertia:

https://github.com/user-attachments/assets/1cc0e743-4052-4507-b6a4-0b1fc515e729

lenroe-nv, 2025-10-08:
I am pretty sure picking is not the cause of this issue, it looks fine to me after fixing the inertia in this PR. [We need to modify ](https://github.com/newton-physics/newton/issues/838) We ](https://github.com/newton-physics/newton/pull/870) 
We need to modify anymal tests though.

vreutskyy, 2025-10-08:
Here's how it looks to me if I only change the inertia in `parse_urdf`
https://github.com/user-attachments/assets/fe94c873-ed98-434d-9c48-e3be15d25e55

lenroe-nv, 2025-10-08:
Agreed that damping for picking is another (preference?) issue. I also think your second video looks more natural. Was not the problem I wanted to show in the issue though, which also persists without picking.